andrivet/ADVi3pp #280: Change of Z offset is not stored in Settings / Sensor UI page

s1mb1o, 2022-03-22:
Thank you for submitting your feedback to the ADVi3++ project.

Please be sure to review our [code of conduct](code_of_conduct.md) and the checklists below before submitting and use one of the templates. 

**It is mandatory**. If you do not comply with these rules, I will **delete** your issue or comment. **I will be very strict**. If you persist, I may **ban** you.

- [x] This issue (or comment) is **NOT** about support. **This is not the right place to ask for support**. 
Instead, go to [ADVI3++ Community - Get Help](https://community.advi3pp.com/c/get-help). 

------------------------------------------------------------
### Bug Report

- [x] You have checked that you are not creating a **duplicate** of an existing (maybe closed) issue (use the search)
- [x] You have tested with the **latest** stable version of ADVi3++
- [x] You are able to reproduce the bug
- [x] You are able to describe step by step the process to reproduce the bug

#### Description of the bug

Manual value change of Z offset in Settings -> Probe UI page is not saved.

* Exact model of your **printer**: Wanhao Duplicator i3 plus mk2 with BLTouch

* Steps to reproduce:
  - Enter Settings -> Sensor page
  - Set value of "Z offset" to -1.0 (or any value below zero)
  - Press (V) in right top corner
  - Enter Settings -> Sensor page
  - "Z offset" still has old value

Issue is in method void SensorSettings::get_values()
Value is read with 
```
uint16_t z = frame.read_word();
```
and then converted to float
```
ExtUI::setZOffset_mm(z / 100.0);
```

Because uint16_t used, we get 65436 value in z variable.
By dividing it by 100.0 we get float 654.36
Which is not in range Z_PROBE_OFFSET_RANGE_MIN ... Z_PROBE_OFFSET_RANGE_MAX
So ExtUI::setZOffset_mm() does not save value.

* Solution:

Use int16_t instead of uint16_t to get negative values.
```
uint16_t z = frame.read_word();
```

andrivet, 2022-03-22:
Duplicate of #268

andrivet, 2022-03-25:
Please avoid creating duplicates. Thanks

closed 2022-03-25 by andrivet