andrivet/ADVi3pp #282: Babystepping display not working

chschmit, 2022-03-28:
### Did you check that your report is not a duplicate?

Yes, I have checked and this is a new bug not previously reported.

### Bug Description

Babystepping display doesn't readout the offset distance...it just shows 0.00

### Expected behavior

With each increase or decrease in babystepping the display would update to show the offset distance

### Actual behavior

With each increase or decrease in babystepping the display is not updated. However, I can visually see the z-axis steppers move more or less with each babystepping up/down button press.

### Steps to Reproduce

1. Go to Print.
2. Go to settings
3. advance the babystepping up/down.
4. Display reads 0.00 regardless of how many up/down button presses.

### Version of ADVi3++ Firmware

5.10

### Printer model

Wanhao Duplicator i3 Plus

### Mods

basic two wire run out sensor

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

andrivet, 2022-03-29:
I am not able to reproduce this behavior. Do you have a sensor such as a BLTouch?

chschmit, 2022-03-30:
No. I don't have a BLTouch installed.

andrivet, 2022-04-24:
I am not able to reproduce this bug and nobody else reported such an issue so far. So I am closing it now. Will reopen it if someone else report the same thing.

closed 2022-04-24 by andrivet