andrivet/ADVi3pp #284: LCD stuck dim when HW error detected

galfisk, 2022-04-21:
### Did you check that your report is not a duplicate?

Yes, I have checked and this is a new bug not previously reported.

### Bug Description

When a HW error is detected (in my case: MINTEMP) while the LCD is dimmed, it doesn't undim when touched, making it harder to read about the error.

### Expected behavior

I expect the LCD to always undim when touched.

### Actual behavior

It doesn't undim when the hardware error message is displayed. 

### Steps to Reproduce

1. Start a printout
2. Wait for the LCD to dim
3. Trigger a hardware malfunction, for instance by disconnecting the hot end thermal sensor 
4. Wait for the printer to show the error screen
5. Try touching the screen to make the LCD undim

### Version of ADVi3++ Firmware

5.0.1

### Printer model

Wanhao Duplicator i3 Plus

### Mods

None

### Bed Leveling

No Bed Leveling

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

I don't know if the LCD undims at the moment when the error is triggered, and then dims again later, or if it stays dim throughout. I haven't found the actual issue triggering this fault during printing, but suspect a bad thermal sensor cable.

andrivet, 2022-04-21:
When a hardware error is detected, the printer is stopped. So there is no way for ADVi3++ to do anything such as un-dimming the LCD panel.

andrivet, 2022-04-21:
The only alternative I see is to un-dimm the LCD panel right before displaying the error. But it means that it will stay like that (un-dimmed) for hours. Not sure other users will agree with that.

galfisk, 2022-04-21:
Personally, I'd like it. It'd make the issue stick out more, especially if you're running several printers at once.
Does the full brightness lead to additional wear on the LCD and/or high power draw, or is dimming a comfort issue only?

andrivet, 2022-04-21:
"Does the full brightness lead to additional wear on the LCD": yes
"high power draw": medium

andrivet, 2022-06-22:
The LCD display will undimm just before displaying the error. The screen also displays which component has failed (if available).

closed 2022-06-22 by andrivet