andrivet/ADVi3pp #285: Show what triggered a HW error

galfisk, 2022-04-21:
### Did you check that your request is not a duplicate?

Yes, I have checked and this is a new request.

### Is your request specific to your setup only or can it benefit to many people?

My request will benefit to many people.

### Is your feature request related to a problem? Please describe.

The feature improves hardware error troubleshooting

### Describe the feature you want

When a MINTEMP error is triggered, it would be useful to know what sensor and temperature triggered the error.

### Additional context

I was trying to troubleshoot a recurring MINTEMP error on a Wanhao Duplicator i3 Plus, which happened some hours into every long printout. I didn't know if it came from bad PID parameters or a bad connection. It would be useful if it said something like MINTEMP EXTRUDER -15°C, then I could deduce that it was a bad connection to the extruder thermistor.

Something similar could probably be useful when trying to pinpoint other hardware issues as well, such as overtemp or homing error.

### Have you provided enough and precise information?

Yes, I have provided precise information in the Additional context box.

andrivet, 2022-04-21:
When a hardware error is triggered, it is dangerous to do anything complex. But maybe, there are some room for improvement here.

galfisk, 2022-05-24:
I realized how you can do this without performing any extra processing after the error has been detected: every time a check for a HW error is performed, a string variable is updated with the pertinent error information. If an error is detected, the checking is halted and the latest string is displayed on the LCD.

andrivet, 2022-05-24:
> the latest string is displayed on the LCD
This is not possible after the firmwares halts and anyway this is not the problem.

andrivet, 2022-06-25:
I have added the temperature that triggers the error on the LCD screen. For other errors (like homing error), it is a little too complex and I don't want to change Marlin's code too much.

closed 2022-06-25 by andrivet