andrivet/ADVi3pp #288: Inversion of black and white BLTouch wires not detected

andrivet, 2022-06-05:
### Did you check that your report is not a duplicate?

Yes, I have checked and this is a new bug not previously reported.

### Bug Description

While installing a BLTouch sensor recently, I found that the ADVi3++ wiring sanity check feature in the firmware validates the wiring as correct even when the black (GND) and white (25/PA3) wires are reversed.

The wiring check passed, but when I homed the print head the extruder nozzle did not stop when the BLTouch probe contacted the bed. It continued to press into the bed until manually powered off.

After some checking and investigation I found that I had reversed the black and white wiring connections for the BLTouch. Reversing them corrected the problem and the BLTouch then functioned normally.

It might be worth making it clearer in the UI that the black/white wiring cannot be fully validated by that check. Or see if maybe there’s a way to improve the check’s logic to detect that wiring issue too?

I’m using a verified genuine BLTouch Smart 3 on a WanHao Duplicator i3 Plus (not 2) with ADVi3++ version 5.2. I wired it per the manual, soldering the headers, using the z-probe pins, and disconnecting the z-stop microswitch. I checked all wires for continuity, checked for shorts between any pins/wires, etc.

### Expected behavior

Would be better if the check is able to detect the problem. If not technically possible, should at least warn the user.

### Actual behavior

It does not detect the inversion.

### Steps to Reproduce

1. Invert BLTouch white and black wires
2. Tuning / Sensor
3. Follow the steps.
4. At the end, everything is green despite the inversion

### Version of ADVi3++ Firmware

5.2.0

### Printer model

WanHao Duplicator i3 Plus

### Bed Leveling

BLTouch (genuine)

ringerc, 2022-06-12:
Note that the black wire is GND, and it's common to the GND used for the VCC/GND red/brown pair.

So I guess the mainboard sensor pin is being pulled to GND via the common ground on the BLTouch when miswired this way, and the BLTouch sensor pin is being pulled to the mainboard's GND.

andrivet, 2022-06-17:
I think I found a (simple) way to detect the inversion. I will test with different versions of BLTouch and if I see no issue, will be part of version 5.4. Closing for the moment

closed 2022-06-17 by andrivet