andrivet/ADVi3pp #289: Travel movements during manual bed levelling significantly slower than earlier versions

cjastone, 2022-06-07:
### Did you check that your report is not a duplicate?

Yes, I have checked and this is a new bug not previously reported.

### Bug Description

When performing manual bed levelling, the speed of travel movements have slowed significantly between my current version (5.2.0) and the previous version I had installed (4.0.6), causing bed levelling to be a much slower process.

This was one of the greatest enhancements above stock firmware - the behaviour has now reverted to similar to that of stock firmware.

### Expected behavior

I expect travel movements to be approximately 50% to 100% faster when performing manual bed levelling.

I expect travel movements to happen in both the X and Y axes to occur simultaneously.

### Actual behavior

Travel movements are around 25 to 50% slower when compared to version 4.0.6 (possibly others).

Travel movements occur independently for each axis; first on the X and second on the Y; under 4.0.6 (and possibly others) these movements would occur simultaneously.  This is partially referenced in #261 - though I note that this also occurs from the initial home position to any point on the bed too, rather than just between levelling points.

The end result of this is the time taken to manually level the bed is around twice as long as under 4.0.6.

### Steps to Reproduce

Move between any two points on the bed in manual levelling mode and observe that the travel speeds are slower than under 4.0.6.

Move between any two points that are both on different positions on both the X and Y axis and observer that the travel between these two points occurs first on the X axis, then on the Y axis.

### Version of ADVi3++ Firmware

5.2.0

### Printer model

Cocoon Create Touch 

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

I understand that it's explicitly requested to not provide video, but the loss of this feature really outweighs the benefits of the new version, I'm likely to return to this version and am happy to provide video if this would be beneficial.

andrivet, 2022-06-07:
Good points, no need for a video, your description is clear enough. I expect to integrate this in the next release.

andrivet, 2022-06-26:
Fixed in version 5.4.0

closed 2022-06-26 by andrivet