andrivet/ADVi3pp #295: Printer thinks it's at 0,0,0 after being powered off

galfisk, 2022-07-25:
### Did you check that your report is not a duplicate?

Yes, I have checked and this is a new bug not previously reported.

### Bug Description

Printer thinks it's at 0,0,0 after being powered off, and consequently the manual "Move" commands don't work correctly. 

### Expected behavior

I expect the X-left, Y-back and Z-down keys to move the corresponding bits until they hit their limit switches. I expect the extruder to move when using the E controls.

### Actual behavior

The X-left, Y-back and Z-down only work after first moving X to the right, Y to the front or Z up, and then only until the corresponding bits reach the positions they had when powering on. The extruder manual control doesn't move the extruder motor in any direction.

### Steps to Reproduce

1. Turn printer off and on while the hot end and bed are in intermediate positions
2. Open the Controls->Move menu
3. Try to move X to the left, Y to the back, or Z downwards.
4. If you "Home" any axis, that axis will start working normally
5. If you perform a "Filament unload", the E minus control starts working
6. If you perform a "Filament load", the E plus control starts working

### Version of ADVi3++ Firmware

5.4.1

### Printer model

Wanhao Duplicator i3 plus, both 5.1 and 5.2c mainboards

### Additional information & file uploads

Tested on 5 printers. 
Could it be #275 with different symptoms?

galfisk, 2022-07-25:
If I try to move too far away form the limit switches before homing, the motors will start to grind, since the printer doesn't understand that it has reached the mechanical end of the axis.

andrivet, 2022-07-25:
This behavior is expected and the way these printers are working since there is no max limit switched. It is not s bug.

galfisk, 2022-07-25:
But wouldn't it make more sense to allow me to move the mechanisms toward the limit switches, and disallow movement away from them? Because right now, I'm prevented from moving towards the switches (which I'd like to do because that would set the proper zero point for the axis), but I am allowed to move too far away from them, putting undue stress on the printer and throwing off the dual Z axis motors.
Also, shouldn't manual extruder motor control be unaffected, since it has no limit switch?

andrivet, 2022-07-25:
Again, this is how those printers are working. When they start, they do not know where the nozzle is so they assume it is at 0,0,0. As far as I know, all similar printers are working this way. You have to home them.

galfisk, 2022-07-25:
Ok, got it.
That still leaves the extruder motor though. What's the assumption that keeps it from moving in any direction using the manual controls?

andrivet, 2022-07-25:
For this, I have to check

galfisk, 2022-07-25:
You could also detect the situation programmatically: if an axis says 0 but its limit switch is not engaged, the firmware could assume that it's not actually 0, and allow manual control of motion towards the switch until it engages. After all, that's pretty much what the home buttons do automatically.

andrivet, 2022-08-01:
I have made some tests to see is something like you said is feasible. It is not. You said "allow manual control of motion towards the switch until it engages". This assumes that the limit switches are monitored all the time. They are not. So instead of grinding on the right, the printer will grind on the left. Limit switches are only used when you home the printer (either from the LCD panel or from a GCode).

So avoiding this problem means disable any movement until you home the printer and I do not want to change this.

So I think there no solution. This is a normal behaviour of those printers. It probably explained also why this is hardcoded in Marlin's code. I am closing this non-issue. I will reopen it if someone has a brilliant idea.

closed 2022-08-01 by andrivet