andrivet/ADVi3pp #302: Updated manual - Link to filament sensor not working

MartynWilliams, 2022-08-06:
### Did you check that your report is not a duplicate?

Yes, I have checked and this is a new bug not previously reported.

### Bug Description

The new manual includes a description of the filament runout screen is informative, however the link describing which sensor to buy and how to connect it to the printer does not work.

Which sensor to buy and how to connect it to the printer is explained in [How to to add a filament runout sensor](https://manual.advi3pp.com/en/touch-panel-guide/settings/runout-settings)

### Expected behavior

Link to a new page showing details of the runout sensor

### Actual behavior

Link returns to the page which called it.

### Steps to Reproduce

Navigate to the runout-settings option in the manual. Clicking on the "which sensor to buy and how to connect it to the printer" does not produce the desired result.

### Version of ADVi3++ Firmware

5.4.2

### Printer model

Not applicable

MartynWilliams, 2022-08-20:
Hi,
I forgot to add the link to the instructions to add the filament sensor.
The link described above should point to this location.
https://manual.advi3pp.com/en/add-runout-sensor

andrivet, 2022-08-20:
Thanks for reporting. Fixed.

closed 2022-08-20 by andrivet