andrivet/ADVi3pp #305: User Manual Site BROKEN

EhRabz, 2022-08-23:
Oops, something went wrong...
Invalid email / username or password.

Even when im 100% , and logged in.

andrivet, 2022-08-23:
You are probably following a wrong link. But you are not telling which one.

closed 2022-08-23 by andrivet

andrivet, 2022-08-23:
I closed it because you provided no information. You are not following the required template.

andrivet, 2022-08-23:
You can't just say "BROKEN", without telling me what it is about, what is broken, etc. I can't guess. So if you provide information and explain exactly what is the problem, I will reopen the issue.

EhRabz, 2022-08-23:
Anytime I go to log in. I get hit with Oops, something went wrong...
Invalid email / username or password. even though the same username and password are used to log into the forum.

andrivet, 2022-08-23:
Are you a sponsor of the project ?

EhRabz, 2022-08-24:
Oh didn't realize its behind a paywall now, but seems like I can't pledge to you for some reason.

andrivet, 2022-08-24:
It is written on the first page: 

Note: Full access to this User Manual is only provided to active sponsors of [the project on Patreon](https://www.patreon.com/bePatron?u=6504486).