andrivet/ADVi3pp #307: Versions over 5.0.3 having issues displaying SD card files

DimitryNechaev, 2022-09-21:
### Did you check that your report is not a duplicate?

Yes, I have checked and this is a new bug not previously reported.

### Bug Description

Hi folks

I've decided to upgrade my version 3 of ADVi3++ to the latest.
I've read through deployment instructions and flashed the latest 5.4.2
To my surprise it had issues accessing SD card. 
In short - it was working correctly with the card it accessed first time after reboot.
Use case 1
- if a card is not plugged in, it will display an empty list when pressing Print -  i.e. it would not detect the there is no card. 
- after plugging a card it will display only the first file from that card
- to see the full list of files I need to reboot the printer (turn off and on)
Use case 2
- if a card was plugged before accessing for the first time, it will display files as expected
- after changing card, it will display files from the previous "session" and won't show new files
- to see updated card content need to reboot.

I started downgrading until I met a first version where card reading worked correctly.
The criteria for correct SD Card reader work is following:
1. When no card plugged in, hitting Print would thing for a bit and write text "no card present" (not exact text, as remembered)
2. When a card is plugged in, accessing the card would return a correct file listing

So the highest functional version is 5.0.3
LCD version does not matter, I tried 5.4.2 LCD with 5.0.3 mainboard version and it worked (with regards to the files)
On the opposite, any mainboard version above 5.0.3 would fail SD Card reader criteria.

I'm surpised that there is nobody reports on the issue, which is so obvious, so I suspected something wrong with my printer.
My printer is Aldi Coocon Create Touch.
It is stock config, i.e. neither equipped with BLTouch nor with filament sensor.

I updated printer with `CleanEEPROM.ino.with_bootloader.mega.hex` after which learned to use avrdude to flash it back.
Unfortunately this exercise didn't solve anything, the issue was still there.
I'm sitting on 5.0.3 and wandering what's going on.

### Expected behavior

The criteria for correct SD Card reader work is following:
1. When no card plugged in, hitting Print would thing for a bit and write text "no card present" (not exact text, as remembered)
2. When a card is plugged in, accessing the card would return a correct file listing

### Actual behavior

Use case 1
- if a card is not plugged in, it will display an empty list when pressing Print -  i.e. it would not detect the there is no card. 
- after plugging a card it will display only the first file from that card
- to see the full list of files I need to reboot the printer (turn off and on)
- 
Use case 2
- if a card was plugged before accessing for the first time, it will display files as expected
- after changing card, it will display files from the previous "session" and won't show new files
- to see updated card content need to reboot.

### Steps to Reproduce

Well, it is as simple as accessing the printer w/o card and then with the card having files on it.

### Version of ADVi3++ Firmware

5.4.2

### Printer model

Aldi Coocon Create Touch

### Mods

None

### Bed Leveling

No Bed Leveling

### Your Slicer

Cura

andrivet, 2022-09-24:
I will do some testing next week

DimitryNechaev, 2022-09-30:
Thanks for confirming it, looking forward for the fix!

andrivet, 2022-10-03:
Fixed (with several other problems related to the SD card) in version 5.5.0

closed 2022-10-03 by andrivet