andrivet/ADVi3pp #308: Update description of Print Settings panel

DimitryNechaev, 2022-09-21:
### Did you check that your request is not a duplicate?

Yes, I have checked and this is a new request.

### Is your request specific to your setup only or can it benefit to many people?

My request will benefit to many people.

### Is your feature request related to a problem? Please describe.

When accessing updated print settings panel I don't understand some functions which are not described in the manual https://manual.advi3pp.com/en/touch-panel-guide/print/print-settings

### Describe the feature you want

The panel has 2 settings.

1. A setting under baby stroller with value 0.04
2. A setting with a switch options 0.02, 0.1 and 1.00

The manual has reference to baby stepping, but it is not clear
- which of 2 options is referred to
- what is a meaning of "baby stepping"
- what is purpose of the other option

Please make it more clear as it is unusable for me at the moment w/o support

### Additional context

https://manual.advi3pp.com/en/touch-panel-guide/print/print-settings

### Have you provided enough and precise information?

Yes, I have provided precise information in the Additional context box.

andrivet, 2022-09-24:
I will add a category for the manual (defect or suggestions). I will look at this soon

andrivet, 2022-10-01:
I will enhance the documentation. In the meantime, here is a description from Marlin’s documentation ([Babystep | Marlin Firmware](https://marlinfw.org/docs/gcode/M290.html)):

> Apply babysteps to one or more axes using current units. Offsets applied with [M290](https://marlinfw.org/docs/gcode/M290.html) aren’t added to the current coordinates, but are intended for making small adjustments, especially in the Z axis, at the start of a print.
>
> Note that when  `BABYSTEP_ZPROBE_OFFSET` is enabled, [M290](https://marlinfw.org/docs/gcode/M290.html) also modifies the Probe Z Offset (with no immediate effects). The new Z offset applies to successive probing operations, and can be saved with [M500](https://marlinfw.org/docs/gcode/M500.html). This behavior is means to coincide with the LCD Menu replacing “Z Babystepping” with “Babystep Z Probe Offset.” To avoid this side-effect, use M290 P0 or leave ?`BABYSTEP_ZPROBE_OFFSET` disabled.
For ADVi3++, `BABYSTEP_ZPROBE_OFFSET` is enabled. Pressing the checkmark triggers a M500.

See also: https://community.advi3pp.com/t/missing-how-to-manually-input-z-offset-information-on-5-4/4386/8

closed 2022-11-02 by andrivet