andrivet/ADVi3pp #315: Beep in Preheat when the target temperature is achieved

andrivet, 2022-10-22:
### Did you check that your request is not a duplicate?

Yes, I have checked and this is a new request.

### Is your request specific to your setup only or can it benefit to many people?

My request will benefit to many people.

### Describe the feature you want

Just emit a beep when the target temperature is achieved to inform the user.

### Additional context

n/a

### Have you provided enough and precise information?

Yes, I have provided precise information in the Additional context box.

andrivet, 2023-07-02:
Implemented in 5.7.0

andrivet, 2023-07-22:
Following tests, the way it is implemented is not right. It beeps too often. So change the implementation.

closed 2023-07-22 by andrivet

Delfana, 2023-10-04:
Can this be turned on/off in printer settings? Or do you have to use the source Configuration files?

andrivet, 2023-10-04:
@Delfana Ask support question on ADVi3++ Community, not here

Delfana, 2023-10-05:
> @Delfana Ask support question on ADVi3++ Community, not here

Well I wasn't really asking for support. I was hoping to contribute to this request since it's still open. But nvm.

andrivet, 2023-10-06:
@Delfana You are asking a question on a Closed issue. I do not see how this can be a contribution. It a support request, so this is not the right place.