andrivet/ADVi3pp #321: Live Z Adjust during print

wnsander232, 2022-11-04:
### Did you check that your request is not a duplicate?

Yes, I have checked and this is a new request.

### Is your request specific to your setup only or can it benefit to many people?

My request will benefit to many people.

### Is your feature request related to a problem? Please describe.

Not a problem.  This is something I miss from a previous version (5.0.2. I believe).  Its icon was a baby carriage.  

### Describe the feature you want

I own six of the I3+ w/ BLTouch mod. After I do a full clean on my printers **Live Z adjustment** saves a lot of time dialing in the first layer as i can make micro-adjustments as I see the layer go down.  Currently in the print settings there is only an indication of Z-Height but no way to adjust it while the print is running.  Now I have to stop the print and go through Z-Height settings again from 0.  Any plans to add this in the future?  I'll likely go back to a previous version as this is something I use quite a bit.

### Additional context

https://projects.ttlexceeded.com/3dprinting_mk3_live_z_calibration.html

### Have you provided enough and precise information?

Yes, I have provided precise information in the Additional context box.

andrivet, 2022-11-04:
Maybe I do not fully understand what you mean but I think that Baby stepping is exactly made for that.

andrivet, 2022-11-04:
https://manual.advi3pp.com/en/touch-panel-guide/print/baby-steps

wnsander232, 2022-11-05:
Hello Andrivet,

Yes! It was right in front of me all along.  Thank you for your great effort and amazing work.

I feel you should begin to expand your printer catalog.  There are many other models that would benefit from your software.  Of course it will expand the headaches as well.

Noah

https://manual.advi3pp.com/en/touch-panel-guide/print/baby-steps

andrivet, 2022-11-05:
>  I feel you should begin to expand your printer catalog

This planned, but it is not the right place to discuss about.

closed 2022-11-05 by andrivet