andrivet/ADVi3pp #349: ADVi3++5 LCD-Panel.sketch element width rendering in Sketch app

taligentx, 2024-09-16:
### Did you check that your report is not a duplicate?

Yes, I have checked and this is a new bug not previously reported.

### Bug Description

At least as of Sketch.app 100.3, some element widths are too small and overflow to two lines - a few sample screens are attached below from the 5.8.0 version of LCD-Panel/Masters/ADVi3++5 LCD-Panel.sketch. 

Interestingly, the same elements render correctly in Lunacy. Typically, the width for these elements is short by 1 pixel and causes an overflow. Font is Avenir Next.

[image]

[image]

[image]

[image]

[image]

[image]

[image]

### Expected behavior

Elements widths should render correctly in the application that originated the .sketch file format.

### Actual behavior

As described.

### Steps to Reproduce

Open ADVi3++5 LCD-Panel.sketch in Sketch.app (macOS).

### Version of ADVi3++ Firmware

5.8.0

### Printer model

n/a

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

andrivet, 2024-09-19:
Thank you for this very detailed bug report. I will do my best to fix those. I am not using Sketch but Lunacy, so some of the problems may be related to that.

andrivet, 2024-10-06:
Fixed in  6.0.0

closed 2024-10-06 by andrivet